NorthernWidget/Project-Apis #7: Which firmware is the default?

awickert, 2020-04-25:
There are three currently. Could we either remove deprecated ones or create a "FirmwareTests" folder or a "tests" subfolder or something like this to distinguish?

bschulz1701, 2020-05-28:
Sorry for the naming confusion, **`LiDAR_InterfaceDemo.ino`** should be the full functional firmware. `LiDAR_Accelerometer_InterfaceDemo.ino` does not have full control functionality for the LiDAR Lite or the hall effect sensor, and `HallEffect_Tester.ino` is simply a demo script to test magnet strength and area of effect most easily

awickert, 2020-05-29:
* Renamed `LiDAR_InterfaceDemo.ino` to `LiDAR_Interface.ino`
* Moved the others to a subfolder with the notes provided here.

closed 2020-05-29 by awickert