NorthernWidget/Project-Apis #8: Which hardware is standard?

awickert, 2020-04-25:
Several versions are listed; could deprecated ones be removed? I have created a v0.0.0 release, and they'll live on in that and in the git history.

bschulz1701, 2020-05-28:
Standard hardware is `LiDAR_Lite_Interface_HardSwitch` the rest exist only as potential designs, but for referance, these are the variations in the designs (Note: **All designs except `LiDAR_Lite_Interface_HardSwitch` are untested!**

`LiDAR_Lite_Interface`
- Most minimal implementation
- Fixed I2C, no UART capability
- No accelerometer 

`LiDAR_Lite_Interface_CAN`
- Unfinished minor variant of `LiDAR_Lite_Interface`
- Uses can type crystal with tabs for `XTAL1` instead of an SMD 
- Does not connately any 'no lead' packages (meaning it could be assembled with just a soldering iron, no hot air)

 `LiDAR_Lite_Interface_HardSwitch`
- A expanded version of `LiDAR_Lite_Interface`
- Adds accelerometer 
- Uses physical switch to toggle between UART and I2C communication (as opposed to being firmware controlled in `LiDAR_Lite_Interface` and `LiDAR_Lite_Interface_CAN`
- Adds power switch to disconnect LiDAR Lite 
- Adds hall effect switch to trigger auto-level or other operation

awickert, 2020-05-29:
Untested designs moved to subdirectories and READMEs added as direct copy/paste from text provided here.

closed 2020-05-29 by awickert